dagoma3d/Marlin-By-Dagoma #11: Update repository with new firmware versions

arthurlutz, 2017-01-29:
Going through http://dist.dagoma.fr/betagoma and http://dist.dagoma.fr/ I can see that a lot of work is being put into the Marlin firmware to customize it to fit the numerous combinations of dagoma printers out there. First of all congrats, it must be a difficult task and the UX of the firmware downloader is very good. 

Would it be possible to have this fork/customization of the https://github.com/MarlinFirmware/Marlin/ repo be done in the open so that users can see the diffs and which versions of Marlin are being used ? It would also enable the community to use more up to date versions of Marlin and maybe backport some of the improvements and customisations to Marlin directly ? 

I think a fork of Marlin which would enable users to see all this information would be a much better open source approach than this repo with copies for given versions of the firmware. Thanks in advance.

dagomafr, 2018-01-24:
Hi Arthur,

Our github account has been left besides for a while, but it is over now. I am about to dust it off.
I will come back to you when this clean step is done.

Regards,
Orel

arthurlutz, 2018-01-24:
Excellent news ! Looking forward to this.

damicreabox, 2018-10-09:
Hi, are you still planning to upgrade the repository with the current version of marlin?

arthurlutz, 2019-01-25:
@dagorei I see that you are working on this repo. Any chance of getting some info here ?

arthurlutz, 2019-01-25:
@erdnaxe seems to be doing some work to do with merging with upstream https://github.com/erdnaxe/Marlin-Discovery200

dagorel, 2019-01-25:
@arthurlutz : Sure.

For now, we have no plan to marge with upstream, but of course it should be nice.
There is some specific stuff for each Dagoma printer that makes the work a bit longer than expected.

I am near to complete it on my own (mostly missing all the gcode_M600() stuff).

arthurlutz, 2019-01-25:
@dagorel thanks for the heads ! Yes it would be nice to be able to contribute to this work !

dagorel, 2019-01-25:
Ok nice! I will push the current state of the merge i did asap (i have some urgent work to do now). Maybe i will use my own fork (https://github.com/dagorel/Marlin), don't know right now but glad to see some potential contributors.

dagorel, 2019-01-25:
Some additional information: This repo was not originall forked from Marlin... That's bad.
That's why i forked it in https://github.com/dagoma3d/Marlin and began minor stuff. I guess i will use this fork to push the merge stuff.

0r31, 2020-06-28:
Hi,

All the migration work is currently gathered in my own Marlin fork : https://github.com/0r31/Marlin
I guess i will make a pull request on the official Marlin repo for the Ultimate (but without the pause button for now).

Regards,
Orel