dagoma3d/Marlin-By-Dagoma #14: Marlin 1.1.9, 2.0

Zezoh, 2019-11-29:
Hi, 

I used your code for FSR with some modifications  to build a custom copy of marlin 1.1.9 and 2.0, I would like to share them with you, version 1.1.9  FSR + MONO fan, tested and it is work like a charm. 
2.0 not tested yet., 

1.1.9 
https://app.box.com/s/dsitr2jbwg5n20s0pqtr7xci9mq3kd1c

2.0 
https://app.box.com/s/m4rcbmt4h6fkng10a73chrtldd2nso5w

arthurlutz, 2019-11-29:
Hi, what is FSR ?

Zezoh, 2019-11-29:
@arthurlutz 

FSR =  force sensing resistor 

This sensor used under the bed of Nava printers

0r31, 2020-06-28:
https://github.com/MarlinFirmware/Marlin/issues/15878

0r31, 2020-06-28:
https://github.com/Zezoh/Marlin-old/commits/bugfix-1.1.x

VincentMasselis, 2021-03-17:
I'm truly interessted about this update of the official Dagoma fw.
Recently I've updated my DE200+screen to a DU+screen+xl setup but the buildtak from the XL add-on is a little bit curved on the Y axis. When y=150, my ideal z-offset will be -3.05 but when y=0 or y=300 it will be -2.95 o -3.00.

By using the lastest marlin firmwares, the E29 command will use more points than 4 points to probe the bed height, it will use at least 6 points. In my case, that will fix my trouble with my bended buildtak since the center of the plate will be measured not only the corners.

arthurlutz, 2021-03-17:
It would be super nice to be able to say and show that dagoma still supports its initial users and customers.

0r31, 2021-03-17:
Hi @VincentMasselis ,

I forked marlin on my own account [here](https://github.com/0r31/Marlin).
I didn't have enough time to implement the main missing feature (play/pause button) but as far as I know, the default branch (disco-ultimate) works pretty fine on the related printer (Disco Ultimate).
This firmware comes with the [UBL feature](https://marlinfw.org/docs/gcode/G029-ubl.html).

As far as i understand : 
- You have a screen (so no need to have the play/pause button working)
- You have the xl addon (minor changes on the configuration has to be made).

Feel free to ask for more information.

Regards,
Orel

InsanityAutomation, 2021-03-17:
This is upstream now for the buttons to configure
`/**
 * User-defined buttons to run custom G-code.
 * Up to 25 may be defined.
 */
//#define CUSTOM_USER_BUTTONS
#if ENABLED(CUSTOM_USER_BUTTONS)
  //#define BUTTON1_PIN -1
  #if PIN_EXISTS(BUTTON1)
    #define BUTTON1_HIT_STATE     LOW       // State of the triggered button. NC=LOW. NO=HIGH.
    #define BUTTON1_WHEN_PRINTING false     // Button allowed to trigger during printing?
    #define BUTTON1_GCODE         "G28"
    #define BUTTON1_DESC          "Homing"  // Optional string to set the LCD status
  #endif
`

Sophist-UK, 2021-03-17:
The Dagoma wiring of the DE200 is non-standard. There is AFAIK no reason why you cannot use the standard Marlin if either:

a. You use the Dagoma Marlin file which describes the Dagoma pinouts for the MKS board; or
b. You change the wiring on your DE200 to the more standard setup for the board.

You could then have all the latest features.

What I am unclear about is what other fixes or enhancements are included in the Dagoma version of Marlin.

It is on my to-do list (but very low down) to submit the special Dagoma DE200 pinout file to the Marlin repo so that it would be  available to all.

InsanityAutomation, 2021-03-17:
Some of those pinouts were already done and files submitted. If there are more that are not there we can go over them and get them caught up very easily.

0r31, 2021-03-17:
@InsanityAutomation, @Sophist-UK ,

Hi guys, glad to read you !

@InsanityAutomation : that's awesome ! I will take a look !

@Sophist-UK : For Disco printers, nothing special about features except the play/pause button (as addon). The main issue with DE200 is it was delivered with different boards so different pinout files -> different configs to be delivered. Moreover, there is all the addon stuff (screen, xl, ...) . For the current firmware provided by Dagoma, all configs/variants are here : https://github.com/dagoma3d/Marlin-By-Dagoma/tree/master/LinuxAddons/Dagoma/conf/serie

0r31, 2021-03-17:
I know i need to handle that, but honestly, it is not part of my priorities for now (but i trie to make it happen).

Each customer can have his solution for his specific config. The main point is to gather it all.

0r31, 2021-03-17:
@Sophist-UK : Correction from my previous comment. The DE200 (DiscoEasy200) was delivered with the very same board (and pinout). Only the very first printer sold by Dagoma was delivered with two different boards (Melzi and MKS).

The remaining stuff is how to handle addons. Could make some arbitrary choices for the default config.

I will discuss with my bosses to check if i can use time next week to deliver the DE200 pinout.

(@InsanityAutomation : if accepted, will do it from my account fork)